engnadeau/pybotics #304: pip is not working

Ali78141, 2018-02-26:
## Expected Behaviour

-

## Actual Behavior

-

## Steps to Reproduce the Problem

1.

## Specifications

```

```

Ali78141, 2018-02-26:
i try pip install pybotics in command line 
i got this message ( pip is not recognized as internal or external command

nnadeau, 2018-02-26:
-  The error message `pip is not recognized as internal or external command` usually means that `pip`  and `Python` are not properly installed on your system
- Try running the following command: `python -V`, what output does it give? If that doesn't work, run `python3 -V`
- Follow this link (https://docs.python.org/3/installing/index.html) on how python packages are installed

Ali78141, 2018-02-26:
I installed python and pycharm and its working fine, i think might be I missed some components or having problem in path setting, i am going to check and see

closed 2018-02-26 by Ali78141

Ali78141, 2018-02-26:
this is path where i installed python  
C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64

nnadeau, 2018-02-26:
While your installation is working, it's currently under your `Visual Studio` install. I'd recommend installing `Python` the standard way: https://www.python.org/downloads/

Your install with then look like this: `C:\Users\nicholas\AppData\Local\Programs\Python\Python36\python.exe`