engnadeau/pybotics #735: Defining a custom robot model [FEATURE]

MCYBA, 2019-11-18:
## User Story

For integrating `pybotics` into my physical robotic structure, I need to use a custom robot model definition.

## Potential Solutions

- A function that read a CSV file that contains D-H parameters, and returns the model Array.
- Instead of CSV files, parameters can be given directly as a list.

nnadeau, 2019-11-21:
@MCYBA 

- if your robot is off-the-shelf (e.g., KUKA, ABB, UR, Fanuc), please make a pull request and add the robot model to [`pybotics/predefined_models.py`](https://github.com/nnadeau/pybotics/blob/master/pybotics/predefined_models.py)
- for a CSV file:

```python
import numpy as np
from pybotics.robot import Robot

# load csv data
my_data = np.genfromtxt('my_file.csv', delimiter=',')

# create robot object
robot = Robot.from_parameters(my_data)
```

nnadeau, 2020-03-06:
@MCYBA how's it going?

nnadeau, 2020-05-17:
Closing stale issue.

closed 2020-05-17 by nnadeau