engnadeau/pybotics #736: [BUG] unable to install the vision in github

Gloriabhsfer, 2019-12-02:
## Description

Hi all, 
when I install the pybotic package through pip. I can install it. But my robot.py file is not same as the version in github. It lack the function

>  from_parameters

`Traceback (most recent call last):
  File "UR5_rl_vimp.py", line 105, in <module>
    robot_ur5 = Robot.from_parameters(ur5())
AttributeError: type object 'Robot' has no attribute 'from_parameters'
QObject::~QObject: Timers cannot be stopped from another thread
`
ps: I modify the parameter in predefined model with UR5 robot. But when I try UR10 which you already have, it still return the error like this.
I think the pip package need to update.
## Steps to Reproduce
1.install pybotics with code 

> pip install pybotics

> pip3 install pybotics

2.import this package in your code:

> from pybotics.robot import Robot
from pybotics.predefined_models import ur5

> robot_ur5 = Robot.from_parameters(ur5())

And then run the code, it will gave you the error like first part. And my robot.py file is in this link:
<script src="https://gist.github.com/Gloriabhsfer/4f44980658851ebb2a952d444d3fe948.js"></script>
comared with the version on github, I think l lack some function, but as I was download with pip, I cannot compile the file with git clone and install manually.

## Additional Context
- Pybotics version: 0.6.0
- python version python3.6
- Dependencies versions ubuntu 18.04

nnadeau, 2019-12-03:
- @Gloriabhsfer, `pybotics` is currently at v0.7.3 on PyPI (https://pypi.org/project/pybotics/)
- Please use `pip install --upgrade pybotics` to upgrade your package to the latest version

Gloriabhsfer, 2019-12-03:
@nnadeau 
Thanks for your reply. I upgrade my pip version to latest, and upgrade the pybotic with: 
`sudo -H pip3 install --upgrade pybotics` My pybotic version is now v0.7.3 
but I still got the error like this:

```
robot_ur5 = Robot.from_parameters(ur5())
AttributeError: type object 'Robot' has no attribute 'from_parameters'
QObject::~QObject: Timers cannot be stopped from another thread
QMutex: destroying locked mutex
```
And the robot.py file doesn't change. Do you have any idea about that? Or can I install this package manually?
Best,
Gloria

nnadeau, 2019-12-03:
@Gloriabhsfer could you please copy and paste your full script, including imports?

Gloriabhsfer, 2019-12-03:
Sure. Because I use other package and I want to simplify it. I do some modification of the example basic_usage.py. The code is in here:
[<script src="https://gist.github.com/Gloriabhsfer/df73c48256d1a69ddcf3df98ca564173.js"></script>](url)
It will still return me the error:

> (py35) gloria@gloria-XPS-13-9370:~/pybotics/examples$ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 31, in <module>
    main()
  File "test.py", line 15, in main
    robot = Robot.from_parameters(ur5())
AttributeError: type object 'Robot' has no attribute 'from_parameters'

I use conda to manage my python environmen. When I upgrade the package, the file in conda environment doesn't change. The robot.py is still looked like this:
<script src="https://gist.github.com/Gloriabhsfer/4f44980658851ebb2a952d444d3fe948.js"></script>

nnadeau, 2019-12-03:
@Gloriabhsfer 
- `conda` doesn't use the same environment/packages as `pip`
- When you perform `pip install`, you're most likely updating the package in pip's package list (open two new terminals and do `pip list` and `conda list` in each, you'll probably get different versions)
- You need activate your conda environment, install pip in that environment, and install pybotics using the pip of that environment: https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-pkgs.html#installing-non-conda-packages

Gloriabhsfer, 2019-12-03:
Thanks for your reply. But I installed my package inside conda environment
This is conda list
`protobuf                  3.4.1            py35he6b9134_0    anaconda
ptyprocess                0.5.2            py35h38ce0a3_0    anaconda
py                        1.8.0                     <pip>
pybotics                  0.6.0                     <pip>
pybullet                  1.8.6                     <pip>
pycparser                 2.18             py35h61b3040_1    anaconda
`
And with pip list

`pyassimp                           3.3        
pybotics                           0.6.0      
pycairo                            1.16.2     
pycococreatortools                 0.2.0      
pycocotools                        2.0.0      
pycparser                          2.19       
pycrypto                           2.6.1      
pydot                              1.2.3      
`
Seems both these package have pybotics been installed. I have activated my environment and use the pip inside my conda environment. It's still I don't have form_parameter function.

I will try uninstall the whole environment and re-install it to see if the issue can be fixed.
Thanks,
Gloria

nnadeau, 2019-12-05:
@Gloriabhsfer both the `conda list` and `pip list` show an old `pybotics` version (0.6.0). 

the `from_parameters` function was added in 0.7.0

Gloriabhsfer, 2019-12-13:
@nnadeau Thank you for your reply, but I still cannot install the right version. I just download the github file and add the absolute directory  inside my code.

closed 2019-12-13 by Gloriabhsfer