engnadeau/pybotics #765: [FEATURE]

kstan79, 2021-02-19:
## User Story
I notice pybotics have build in forward and inverse kinematic method and I wish to implement in robot arm. However the prepare documentation i can't understand. Is it possible I supply length of each joint, limits, and suitable parameter so I can generate joint angle by define xyz position?

nnadeau, 2021-02-19:
@kstan79 yes, but you need to define the [Modified Denavit–Hartenberg parameters](https://en.wikipedia.org/wiki/Denavit%E2%80%93Hartenberg_parameters#Modified_DH_parameters) (MDH) of your robot arm.

- MDH parameters have four values: `alpha`, `a`, `theta`, `d`
- your `a` and `d` parameters define linear length

for example, this is what a [basic UR10 robot arm](https://www.universal-robots.com/products/ur10-robot/) looks like as MDH parameters: 

```python
ur10 = np.array(
    [
        [0, 0, 0, 118],
        [np.pi / 2, 0, np.pi, 0],
        [0, 612.7, 0, 0],
        [0, 571.6, 0, 163.9],
        [-np.pi / 2, 0, 0, 115.7],
        [np.pi / 2, 0, np.pi, 92.2],
    ]
)
```

if we wanted to make a section of the robot longer/shorter, we would modify the `a` or `d` parameters in this kinematic description.

now, for one of your direct questions, `generate joint angle by define xyz position?`, we use the forward kinematics function, `fk()`:

```python
import numpy as np
from pybotics.geometry import matrix_2_vector
from pybotics.predefined_models import ur10
from pybotics.robot import Robot

# prevent scientific notation when printing
np.set_printoptions(suppress=True)

# create robot model
robot = Robot.from_parameters(
    np.array(
        [
            [0, 0, 0, 118],
            [np.pi / 2, 0, np.pi, 0],
            [0, 612.7, 0, 0],
            [0, 571.6, 0, 163.9],
            [-np.pi / 2, 0, 0, 115.7],
            [np.pi / 2, 0, np.pi, 92.2],
        ]
    )
)

# get the forward kinematics when all joints = 0
joints = [0, 0, 0, 0, 0, 0]
pose = robot.fk(joints)

print(f"Joints:\n{joints}")
print(f"Pose as a matrix:\n{pose}")
print(f"Pose as a vector:\n{matrix_2_vector(pose)}")
print('\n\n')

# get the forward kinematics after moving first joint 10deg
joints = np.deg2rad([10, 0, 0, 0, 0, 0])
pose = robot.fk(joints)

print(f"Joints:\n{joints}")
print(f"Pose as a matrix:\n{pose}")
print(f"Pose as a vector:\n{matrix_2_vector(pose)}")
```

```
Joints:
[0, 0, 0, 0, 0, 0]
Pose as a matrix:
[[    1.      0.      0.  -1184.3]
 [   -0.      0.     -1.   -256.1]
 [   -0.      1.      0.      2.3]
 [    0.      0.      0.      1. ]]
Pose as a vector:
[-1184.3         -256.1            2.3           -0.
     0.             1.57079633]

Joints:
[0.17453293 0.         0.         0.         0.         0.        ]
Pose as a matrix:
[[    0.98480775     0.             0.17364818 -1121.83652359]
 [    0.17364818     0.            -0.98480775  -457.86080236]
 [   -0.             1.             0.             2.3       ]
 [    0.             0.             0.             1.        ]]
Pose as a vector:
[-1121.83652359  -457.86080236     2.3            0.17453293
     0.             1.57079633]
```

nnadeau, 2021-02-19:
similarly, we use inverse kinematics, `Robot.ik()`, to go from XYZ -> joints

kstan79, 2021-02-20:
thanks of your reply, I'm studying the official UR10 parameter seems very different with your parameter.

https://www.universal-robots.com/articles/ur/application-installation/parameters-for-calculations-of-kinematics-and-dynamics/
```
alpha,         a,      theta,  d
 np.pi/2,       0,      0,      0.1273
 0,         -0.612,     0,      0
 0,         -0.5723,    0,      0
 np.pi/2,   0,          0,      0.163941
 -np.pi/2,  0,          0,      0.1157
 0,         0,          0,      0.0922
 ```

I'm know some involve conversion from meter to mm, but the dh parameter look very different. Example: Joint 1. Is it we have to modified something?

nnadeau, 2021-02-22:
hi @kstan79 , 

UR's website uses standard DH parameters. Pybotics uses the Modified DH parameters (https://en.wikipedia.org/wiki/Denavit%E2%80%93Hartenberg_parameters#Modified_DH_parameters) as introduced in  `John J. Craig, Introduction to Robotics: Mechanics and Control (3rd Edition) ISBN 978-0201543612`

nnadeau, 2021-02-22:
also, `matrix_2_vector()` (https://pybotics.readthedocs.io/en/latest/pybotics.geometry.html#pybotics.geometry.matrix_2_vector) converts a 4x4 transform matrix to a vector of [X, Y, Z, rZ, rY, rX], whereby the last three values are EULER ZYX rotations

nnadeau, 2021-02-22:
@kstan79 i highly recommend reading  `John J. Craig, Introduction to Robotics: Mechanics and Control (3rd Edition) ISBN 978-0201543612`; you'll be an expert in no time

kstan79, 2021-02-23:
Thanks nnadeau your advise, will find time to read the book

By the way I'd tried as your advise, the result `robot.ik` result is None:

```
kstan@kstans-MacBook-Pro ar3_webservice % /usr/local/bin/python3 test.py
Prev vector:
[369.2        -16.61        -0.           3.14159265   1.51460937
  -1.57079633]
New vector:
[374.2        -11.61        -0.           3.14159265   1.51460937
  -1.57079633]

Prev matrix:
[[ -0.0561574    0.99842193  -0.         369.2       ]
 [  0.          -0.          -1.         -16.61      ]
 [ -0.99842193  -0.0561574   -0.          -0.        ]
 [  0.           0.           0.           1.        ]]
New Matrix:
[[ -0.0561574    0.99842193  -0.         374.2       ]
 [  0.          -0.          -1.         -11.61      ]
 [ -0.99842193  -0.0561574   -0.          -0.        ]
 [  0.           0.           0.           1.        ]]

Prev Joints:
[0, -90, 1.05, 0, 0, 0]
New joints rad:
None
```
Below is my sample source code:
```
import numpy as np
from pybotics.geometry import matrix_2_vector,vector_2_matrix
from pybotics.predefined_models import ur10
from pybotics.robot import Robot

# prevent scientific notation when printing
np.set_printoptions(suppress=True)

# create robot model

dhparas=    np.array([
            #alpha,            a,      theta,      d
            [-np.pi/2,         64.2,   0,          169.77],
            [0,                305,    0,      0],
            [np.pi,             0,    -np.pi/2,          0],
            [-np.pi,            0,      0,          -222.63],
            [np.pi,             0,      0,          0],
            [0,                 0,      np.pi,      -36.25],
        ])

robot = Robot.from_parameters(dhparas)
# get the forward kinematics when all joints = 0
joints = [0, -90, 1.05, 0, 0, 0]
pose = robot.fk(joints)
newvectorpose = matrix_2_vector(pose)
newvectorpose[0] = newvectorpose[0]+5
newvectorpose[1] = newvectorpose[1]+5

print(f"Prev vector:\n{matrix_2_vector(pose)}")
print(f"New vector:\n{newvectorpose}")
print('\n\n')

newmatrix =  vector_2_matrix(newvectorpose)
newjointsrad = robot.ik(newmatrix)

print(f"Prev matrix:\n{pose}")
print(f"New Matrix:\n{newmatrix}")
print('\n\n')
print(f"Prev Joints:\n{joints}")
print(f"New joints rad:\n{newjointsrad}")
```

Am I miss some code before `robot.ik()`?

nnadeau, 2021-02-24:
@kstan79 the joints need to be in radians, use `np.deg2rad()` to convert `joints = [0, -90, 1.05, 0, 0, 0]`

kstan79, 2021-02-24:
thanks reply, the result same:

```
kstan@kstans-MacBook-Pro ar3_webservice % python3 test.py 
Prev vector:
[369.2        -16.61        -0.           3.14159265   0.01832596
  -1.57079633]
New vector:
[374.2        -11.61        -0.           3.14159265   0.01832596
  -1.57079633]

Prev matrix:
[[ -0.99983208   0.01832493  -0.         369.2       ]
 [  0.           0.          -1.         -16.61      ]
 [ -0.01832493  -0.99983208  -0.          -0.        ]
 [  0.           0.           0.           1.        ]]
New Matrix:
[[ -0.99983208   0.01832493  -0.         374.2       ]
 [  0.          -0.          -1.         -11.61      ]
 [ -0.01832493  -0.99983208   0.          -0.        ]
 [  0.           0.           0.           1.        ]]

Prev Joints:
[ 0.         -1.57079633  0.01832596  0.          0.          0.        ]
New joints rad:
None
kstan@kstans-MacBook-Pro ar3_webservice % clear
kstan@kstans-MacBook-Pro ar3_webservice % clear
kstan@kstans-MacBook-Pro ar3_webservice % clear
kstan@kstans-MacBook-Pro ar3_webservice % python3 test.py
Prev vector:
[369.2        -16.61        -0.           3.14159265   0.01832596
  -1.57079633]
New vector:
[374.2        -11.61        -0.           3.14159265   0.01832596
  -1.57079633]

Prev matrix:
[[ -0.99983208   0.01832493  -0.         369.2       ]
 [  0.           0.          -1.         -16.61      ]
 [ -0.01832493  -0.99983208  -0.          -0.        ]
 [  0.           0.           0.           1.        ]]
New Matrix:
[[ -0.99983208   0.01832493  -0.         374.2       ]
 [  0.          -0.          -1.         -11.61      ]
 [ -0.01832493  -0.99983208   0.          -0.        ]
 [  0.           0.           0.           1.        ]]

Prev Joints rad:
[ 0.         -1.57079633  0.01832596  0.          0.          0.        ]
New joints rad:
None
kstan@kstans-MacBook-Pro ar3_webservice % 

```

the source code i'd modified as below:
```
import numpy as np
from pybotics.geometry import matrix_2_vector,vector_2_matrix
from pybotics.predefined_models import ur10
from pybotics.robot import Robot

# prevent scientific notation when printing
np.set_printoptions(suppress=True)

# create robot model

dhparas=    np.array([
            #alpha,            a,      theta,      d
            [-np.pi/2,         64.2,   0,          169.77],
            [0,                305,    0,      0],
            [np.pi,             0,    -np.pi/2,          0],
            [-np.pi,            0,      0,          -222.63],
            [np.pi,             0,      0,          0],
            [0,                 0,      np.pi,      -36.25],
        ])

robot = Robot.from_parameters(dhparas)
# get the forward kinematics when all joints = 0
joints = np.deg2rad([0, -90, 1.05, 0, 0, 0])

pose = robot.fk(joints)
newvectorpose = matrix_2_vector(pose)
newvectorpose[0] = newvectorpose[0]+5
newvectorpose[1] = newvectorpose[1]+5

print(f"Prev vector:\n{matrix_2_vector(pose)}")
print(f"New vector:\n{newvectorpose}")
print('\n\n')

newmatrix =  vector_2_matrix(newvectorpose)
newjointsrad = robot.ik(newmatrix)

print(f"Prev matrix:\n{pose}")
print(f"New Matrix:\n{newmatrix}")
print('\n\n')
print(f"Prev Joints rad:\n{joints}")
print(f"New joints rad:\n{newjointsrad}")
```

kstan79, 2021-02-24:
i notice i comment below code, the calculation seems working and it inverse back to original joint value value :
```
pose = robot.fk(joints)
#newvectorpose[0] = newvectorpose[0]+5
#newvectorpose[1] = newvectorpose[1]+5
```
I have any mistake? I tried change `+5` to `5.`,`1` it seems any modification will cause  on vector value will cause result = `None`

kstan79, 2021-03-02:
Just follow up in case didn’t notice my reply. Thanks

nnadeau, 2021-03-02:
so this is the fun part of robotics lol, inverse kinematics do not necessarily have a closed-form solution. If you look at the source for `ik()`, you'll notice that I cheated and used a least-squares minimization as a "general" solution to the IK problem. 

https://github.com/nnadeau/pybotics/blob/90e3503417300fc63b2f224a416c1f95369dc9b3/pybotics/robot.py#L84-L97

This is often good enough for a lot of cases, but you may have come across a condition where it didn't work. Note, that your joints are very close to a singularity which is the enemy of robotics and robust Cartesian movements.

Here's an example with the UR10 and "very bent" joints to avoid singularities.

```python
import numpy as np
from pybotics.geometry import matrix_2_vector, vector_2_matrix
from pybotics.predefined_models import ur10
from pybotics.robot import Robot

np.set_printoptions(suppress=True)

robot = Robot.from_parameters(ur10())

joints = np.deg2rad([10, -20, 30, -40, 50, -60])
pose = robot.fk(joints)

print(f"Original joints:\n{joints}")
print(f"Original pose:\n{pose}")

pose[0, -1] += -5
joints = robot.ik(pose, q=joints)

print(f"New pose:\n{pose}")
print(f"New joints:\n{joints}")
```

```
Original joints:
[ 0.17453293 -0.34906585  0.52359878 -0.6981317   0.87266463 -1.04719755]
Original pose:
[[   -0.08581649     0.83616923    -0.5417163  -1199.82332389]
 [   -0.40406272    -0.52620898    -0.74822284  -438.16892181]
 [   -0.9106969      0.1546775      0.38302222   163.41395307]
 [    0.             0.             0.             1.        ]]
New pose:
[[   -0.08581649     0.83616923    -0.5417163  -1204.82332389]
 [   -0.40406272    -0.52620898    -0.74822284  -438.16892181]
 [   -0.9106969      0.1546775      0.38302222   163.41395307]
 [    0.             0.             0.             1.        ]]
New joints:
[ 0.17381232 -0.33326577  0.49167304 -0.68242807  0.8720764  -1.04665002]
```

If you're working with a specific robot, i recommend overriding the `ik()` function with an IK solution specific to the robot.

kstan79, 2021-03-12:
ah,Im building using AR3 (https://www.anninrobotics.com/), it is nice project and I wish to rewrite their source code to allow it run as web services. 

The basic movement I'd manage to migrate into new web service (https://github.com/SIMITGROUP/ar3_robotarm_webservice), but its kinematic calculation look too tedious and confusing, so I wish to find alternative library and redo it. You have any recommendation to do ik beside this?

nnadeau, 2021-03-12:
@kstan79 to be honest, IKs aren't usually easy. It's really either a mathematical closed-form solution (i.e., a formula) or some optimization method. If you want to be really trendy, I've seen some recent papers about model-less neural network kinematics :man_shrugging:

kstan79, 2021-03-13:
ok tq

closed 2021-03-13 by kstan79