engnadeau/pybotics #788: Project Idea

adityaanirudhk, 2022-08-14:
Hey I'm working a project on the topic "COMPLETELY AUTOMATED SPACE ROVER USING ARTIFICIAL NEURAL NETWORKS" will you be able to suggest the algorithms which I'll be able to use for my project from this repository. Thank you.

engnadeau, 2022-08-15:
hi @adityaanirudhk !

Given the vagueness of the topic's title, I'm not sure what algorithms you would need. If you need to calibrate any effectors on the rover, then the calibration modules would be best.

adityaanirudhk, 2022-08-15:
I apologize for not giving a brief idea about my idea. I actually am trying to find algorithms for trajectory optimization using either spline theory/De-Castelju's theorems. So is there anything that I can find in this repository related to path planning? 

And also thanks for your suggestion on the calibration module. That will help me.

engnadeau, 2022-08-16:
no, i don't have those theorems in this package. Feel free to contribute and add them! :)

closed 2022-08-16 by engnadeau